stlehmann/pyads #453: Reading/Writing of T_MaxString doesn't work properly

joellutz, 2025-04-08:
I have some [Tc2_System.T_MaxString](https://infosys.beckhoff.com/index.php?content=../content/1031/tcplclib_tc2_system/31062795.html&id=) variables I want to read and write. With `plc.read_by_name()` / `plc.write_by_name()` this works only if the `plc_datatype` is specified. Otherwise I get an exception. Similar issue when using `plc.get_symbol()`. There I have to manually set the `plc_type` of the symbol, in order to get it to work.

Because I have to read/write various different variable types dynamically, I would prefer not to specify the `plc_datatype` / `symbol.plc_type` manually. Is there a way to fix this?

Here some details:

```python
import pyads
plc = pyads.Connection('199.4.42.250.1.1', pyads.PORT_TC3PLC1)
plc.open()
symbol_name_string = 'MyStringValue'
plc.read_by_name(symbol_name_string, plc_datatype=pyads.PLCTYPE_STRING) # works
plc.read_by_name(symbol_name_string) # TypeError: 'NoneType' object is not callable
```

```
Traceback (most recent call last):
  File "c:\Temp\pyads_string_test.py", line 9, in <module>
    plc.read_by_name(symbol_name_string) # TypeError: 'NoneType' object is not callable
  File "C:\Users\User\AppData\Roaming\Python\Python310\site-packages\pyads\connection.py", line 546, in read_by_name
    return adsSyncReadByNameEx(
  File "C:\Users\User\AppData\Roaming\Python\Python310\site-packages\pyads\pyads_ex.py", line 1216, in adsSyncReadByNameEx
    value = adsSyncReadReqEx2(
  File "C:\Users\User\AppData\Roaming\Python\Python310\site-packages\pyads\pyads_ex.py", line 845, in adsSyncReadReqEx2
    data = data_type()
TypeError: 'NoneType' object is not callable
```

```python
symbol = plc.get_symbol(symbol_name_string, plc_datatype=pyads.PLCTYPE_STRING)
print(f'{symbol.symbol_type}') # Tc2_System.T_MaxString
print(f'{symbol.plc_type}') # None
print(f'symbol {symbol_name_string} = {symbol.read()}') # TypeError: 'NoneType' object is not callable

symbol = plc.get_symbol(symbol_name_string)
symbol.plc_type = pyads.PLCTYPE_STRING
print(f'symbol {symbol_name_string} = {symbol.read()}') # works
```

chrisbeardy, 2025-04-09:
Thanks for the issue report, will have to investigate this, in the meantime, might be worth going back a version (or two) if you don't need the fixes/functionailty of these versions while we investigate. Some string changes were made in the latest updates.

joellutz, 2025-05-07:
Addition: When the variable is a property of type `T_MaxString` (with [monitoring pragma](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_plc_intro/2529692299.html&id=6668617382077000998)) in the PLC:
``` st
{attribute 'monitoring' := 'call'}
PROPERTY MyMAXSTRINGProperty : T_MaxString
```
Reading the property works (currently only with `read_by_name` as mentioned in #261), but writing does _not_ work:
``` python
import pyads
plc = pyads.Connection('199.4.42.250.1.1', pyads.PORT_TC3PLC1)
plc.open()
symbol_name_string = 'MyMAXSTRINGProperty'
plc.read_by_name(symbol_name_string, plc_datatype=pyads.PLCTYPE_STRING) # works
plc.write_by_name(symbol_name_string, "Hello World", plc_datatype=pyads.PLCTYPE_STRING) # ADSError: parameter size not correct (1797).
```

```
Traceback (most recent call last):
  File "c:\Temp\property_test.py", line 7, in <module>
    plc.write_by_name(symbol_name_string, "Hello World", plc_datatype=pyads.PLCTYPE_STRING)
  File "C:\Program Files\Python310\lib\site-packages\pyads\connection.py", line 692, in write_by_name
    return adsSyncWriteByNameEx(
  File "C:\Program Files\Python310\lib\site-packages\pyads\pyads_ex.py", line 1257, in adsSyncWriteByNameEx
    adsSyncWriteReqEx(port, address, ADSIGRP_SYM_VALBYHND, handle, value, data_type)
  File "C:\Program Files\Python310\lib\site-packages\pyads\pyads_ex.py", line 655, in adsSyncWriteReqEx
    raise ADSError(error_code)
pyads.pyads_ex.ADSError: ADSError: parameter size not correct (1797).
```

This is regardless if i multiply the `plc_datatype` with 255 or any other number.