stlehmann/pyads #456: write_list_by_name Function Produces ADS Error 1793 When Writing to EtherCAT I/Os

AskicV, 2025-05-04:
I'm not entirely sure whether this is expected behavior or an actual issue, but it appears to be the latter.

I'm trying to read from and write to EtherCAT device I/Os. The ADS server is already enabled in the Image settings.
The functions write_by_name, read_by_name, and read_list_by_name work without any problems.

However, when I try to use write_list_by_name—even when passing the exact dictionary returned by a previous read_list_by_name call—I receive error 1793, indicating that the operation is not supported by the ADS server.
It doesn't seem logical that reading a list is supported, but writing a list is not.

Here is how the error can be reproduced:
`import pyads
from pyads.constants import PLCTYPE_BOOL

AMS_NET_ID = "192.168.1.65.1.1"
DEVICE_PORT = 27905

if __name__ == "__main__":

    with pyads.Connection(AMS_NET_ID, DEVICE_PORT) as plc:
        # This variable name exists (read from HW tree)
        var_list = ["Term 2 (EL1008).Channel 1.Input"]

        read_value = plc.read_list_by_name(var_list)
        #  The function returns {'Term 2 (EL1008).Channel 1.Input': False}
        print(read_value)
	  # plc.write_list_by_name(read_value) # Causes ADS error 1793
	  # Create new var dict
        var_to_write = {"Term 2 (EL1008).Channel 1.Input": False}
        # plc.write_list_by_name(var_to_write)  # Causes ADS error 1793
        
        plc.write_by_name("Term 2 (EL1008).Channel 1.Input", True, PLCTYPE_BOOL)  # OK`

SpareSimian, 2025-05-07:
I'm seeing the same error reading axis parameters (ID, name) from an EtherCAT drive. I'm wondering if the EtherCAT ADS server doesn't recognize the newer ADS APIs like adsSyncReadReqEx2.
```
axisId = 1
actualAxisId = plc.read(0x4000 + axisId, 0x00000001, pyads.PLCTYPE_UDINT)
```
Error:
```
Traceback (most recent call last):
  File "C:\devel\pyads\pyads-test.py", line 8, in <module>
    actualAxisId = plc.read(0x4000 + axisId, 0x00000001, pyads.PLCTYPE_UDINT)
  File "C:\Python\Python313\Lib\site-packages\pyads\connection.py", line 370, in read
    return adsSyncReadReqEx2(
        self._port,
    ...<5 lines>...
        check_length,
    )
  File "C:\Python\Python313\Lib\site-packages\pyads\pyads_ex.py", line 864, in adsSyncReadReqEx2
    raise ADSError(error_code)
pyads.pyads_ex.ADSError: ADSError: Service is not supported by server (1793). 
```

SpareSimian, 2025-05-07:
Alas, that's not the problem for my case. A quick test of the C function AdsSyncReadReq fails with the same error code.

SpareSimian, 2025-05-07:
Aha. What did work was to use port 500, which is what the NC system listens on: https://infosys.beckhoff.com/english.php?content=../content/1033/tcplclib_tc2_system/31064331.html

AskicV, 2025-05-14:
I have contacted Beckhoff support and they confirmed that TwinCAT EtherCat Master only supports Read-AdsSumCommands and not Write-AdsSum.
So this has nothing to do with pyads and this issue can be closed.

closed 2025-05-14 by AskicV

chrisbeardy, 2025-05-14:
thanks for investigating this